RoCorbera/BlueVGA #1: Are potential divider resistors required on VGA pins?

souravg009, 2020-05-15:
Hi I was going through your project and in the stm32duino thread, (https://stm32duinoforum.com/forum/viewtopic_f_2_t_3318.html). Amazing work! Just out of curiocity, are any resistors required between R,G,B pins of VGA and STM32 mcu? Other 'works' with VGA have several "potential divider" resistors of 220 Ohm as the pins MAX input voltage is ~1V. I am about to walk on your path, so I thought, its better to ask you before damaging something. :innocent:

RoCorbera, 2020-05-15:
Hi,
Actually the thread for this project is at https://www.stm32duino.com/viewtopic.php?f=10&t=347&sid=4dd9fdf3c73c07ed2ba396bda4120798

You can use a 330 ohms resistor like this: 
STM32 pin ---> 330 ohms ---> connection to VGA pin.

Doing so when STM32 pin is high, the voltage will drop from 3.3V to 0.6V.

When using modern LCD VGA monitors, I use to connect STM32 directly to VGA and I never had any issues... But to be safer just use a 330 ohms resistor as I wrote.

I hope it works fine for you!

RoCorbera, 2020-05-16:
Closing it, as this is not an issue related to the library code.

RoCorbera, 2020-11-24:
Keeping this issue open for consulting similar questions.