RoCorbera/BlueVGA #11: multiple definition of `TIM1_CC_IRQHandler';

djnaoki-mrn, 2022-11-20:
hi, i would like to use your pretty cool library in order to make a monitor of my car parking sensors but i've got some issues.
first, i installed the library from the new arduino ide 2.0.2 library manager and when i tryed to compile the ''hello world'' sketch i got an issue with ''bluebitmap.cpp'' it seems to have a issue with this line : 
''uint32 color32Bits = color << 24 | color << 16 | color << 8 | color;''

i simply replaced with : 
''uint32_t color32Bits = color << 24 | color << 16 | color << 8 | color;''

the second issue is that i have this error while compiling : ''multiple definition of `TIM1_CC_IRQHandler';''

have you an idea of how i can fit it ?

i'm working with STM32F103C8T6 (chinese version) and official STM32 Core v2.3.0 (lastest version). also, i'm using HID 2.0 bootloader (lastest version).

thank you.

RoCorbera, 2022-12-15:
This issue is due to changes in the version 2.3.0.
It has broken the use of `build_opt.h` for, at least Windows, that had the option to disable the HardwareTimer.
Therefore, there is a conflict of function redefinition.

Are you using Windows, Mac or Linux?

RoCorbera, 2022-12-15:
This issue is due to a regression in v2.3.0 as explained in https://github.com/stm32duino/Arduino_Core_STM32/issues/1837#issuecomment-1261884640

There is a way to solve it as explained in the link above:
```
there is a regression with 2.3.0 using the build_opt.h.

It is fixed in the main branch and will be available with the next release 2.4.0.
You could simply replace the prebuild.sh script by this file one:
https://github.com/stm32duino/Arduino_Core_STM32/blob/main/system/extras/prebuild.sh
```